automatedchemistry/flowchem #145: Implementation of a "monitoring of data" as watchdog in the flowchem

samuelvitorsaraiva, 2024-09-11:
The idea is to implement a class where the datas - input and outputs of the api-nodes- is monitoring in order to create a datatable to the experimental run.

JB-Wolf, 2024-09-17:
In my opinion, this is how we should do it:
- expose logging as a component like in helao: 
https://onlinelibrary.wiley.com/doi/10.1002/admi.202101987
- given the toml (for now) by inspection one can find all devices that read data, so that inherit from Sensor
- log ALL sensors together with the device ids/Adresses for linking
- If one exposes that virtual device (one logger for all sensors) as endpoint, You can start and stop logging, and provide argument of where it should be logged to and as what, so to define file/database target resource and basically a pointer, eg experiment id.
- the watchdog could be attached to each sensor on sensor level, I think this is easier for the user and for future use
-  the watchdog values and units than can be collected in the logger and both, logging and watchdog, be done in one background process
-  on the same virtual component (logging component) out of spec reading could then set an alarm with info on which sensor and why to a dedicated endpoint
-  one could even think about to not only put safety watchdog, but also target reached in the same background process

The Idea behind that would be to achieve a very simple orchestration client on top

samuelvitorsaraiva, 2024-09-17:
Well, there are two ways to do it.

Method 1 involves the creation of a virtual device. This device inspects all server devices and generates a log. The process is initiated through an automatic decorator insertion in the FlowchemComponent, and it functions effectively. I did it.

2 - Run another "client," which will inspect the server and create a loop to trigger all components' get methods and save the datas. It is more elegant, but I am working on it now to see how complex it will be.