automatedchemistry/flowchem #150: FastAPI isn't handling well not basic type arguments: List, Dict, ...

samuelvitorsaraiva, 2024-09-27:
I was interacting with a Knauer Valve using the ComponentClient and trying to send a list (e.g., [[4, 0]]) as a parameter to a FastAPI endpoint. The issue arises because requests encodes the list in a format that FastAPI is not able to handle correctly.

Knauer Valve 16 position, toml file:
[device.test]
mac_address = "00:80:A3:CE:7E:CB"
network = "141.14.*.*"

The failure occurs because FastAPI does not correctly parse this complex data format in query parameters by default.

The proposed solution is to inspect the params argument before making the request and, if necessary, serialize the list into a string manually.

cynthiabour, 2025-01-21:
I don't have this issue at all. please provide the error message for evaluation

samuelvitorsaraiva, 2025-01-24:
This problem was solved by a pull request linked to. More details about the problem can be found there. More details about the problem can be found there.

closed 2025-01-24 by samuelvitorsaraiva