automatedchemistry/flowchem #161: The FlowchemComponent class should be base on ABC class

samuelvitorsaraiva, 2025-02-07:
Many methods from the component layer need to be overridden in the device layer. Using the Abstract Base Class with the @abstractmethod decorator would be very useful. What you guys think? @cynthiabour and @JB-Wolf?

JB-Wolf, 2025-02-09:
Absolutely, I think I created an issue for that already, please check. Otherwise there should be, but only for required component functions.
We need granularity, some need to be overridden, and some can be, and if not functional.they must throw an error

samuelvitorsaraiva, 2025-03-11:
Another's issues addressed in this branch:

Uniformity of Controllers and Sensors:

    The controller could inherit from the sensor for better consistency.
    Currently, the controller returns values in millibar while the sensor returns in a specified unit. Review and ensure uniformity across similar components.

Zero Bar Initialization:

    The EPC device is initialized to 0 bar, but it is unclear why this is necessary. Investigate and document the reasoning.

Design Principle Violation:

    The current implementation defines behavior directly within the component/device implementation, which violates the design principle. Reconsider the approach and create a more modular or uniform design.

EPC Device Inheritance Issue:

    The EPC device only inherits from Sensor. Investigate why there is no Controller object and determine whether creating a separate controller object is necessary.

Potential Duplication of Lamp State Logic:

    The set_lamp method with states on and off might be duplicating functionality. Analyze whether this method is redundant or can be simplified.

samuelvitorsaraiva, 2025-03-12:
The lamp have not a dedicated status function of KnauerDAD

samuelvitorsaraiva, 2025-03-13:
fast api does not identify pint unit!
components/technical/pressure
async def set_pressure(self, pressure: str) -> pint.Quantity: