automatedchemistry/flowchem #170: Daisy chained ML600 cannot be initialized parallelly.

magz97, 2025-04-10:
File: core.py
Line: 88

Currently, initialize() is run in parallel for all hardware devices using:

`await asyncio.gather(*[dev.initialize() for dev in self.devices])
`
However, this causes issues when initializing multiple ML600 devices simultaneously, as they appear to share a communication port and conflict during setup.

Running them sequentially like this:
```
for dev in self.devices:
    await dev.initialize()
```
resolves the issue, but increases initialization time.

We could consider adding logic to detect daisy-chained or port-sharing devices (like multiple ML600 units), and handle their initialization sequentially, while allowing others to still be initialized in parallel.

JB-Wolf, 2025-04-10:
Great! The underlying problem is identified - please identify what each device exactly does on init: Mostly they return to the 0-position, since this can pump contents into ports they shouldnt go or fail due to high bpr, that kind of init should not be run automatically anyway

JB-Wolf, 2025-04-10:
Additionally, your solution is good - additional, flexible waiting time in case of daisy chained equipment is finre. since they share a port, this could be a handle to initialise all devices on one port sequentially per default?